open-dynamic-robot-initiative/master-board #1: Computer's Mac address should not be hardcoded

thomasfla, 2019-07-28:
https://github.com/solo-bl/master-board/blob/b9e5ec4361b382b84c0e68902e9fa9af10b36f29/software/scripts/src/main.cpp#L13

We should instead get the host mac address from a system call given the interface name.

paLeziart, 2019-11-29:
@thomasfla Is is still required or can we close the issue? The test bench works without changing the Mac address so something may have been modified in the code and the address is not required anymore.

thomasfla, 2019-11-29:
The master board is configured to accept packets from any MAC sources at the moment. But MAC standard still impose us to provide a source address. It's a bit dirty and make the network analysis difficult to keep an hard coded value here.

Also, later, it could be useful to filter incoming packets in the Master Board when using WiFi.