open-dynamic-robot-initiative/master-board #3: Firmware incompatible with latest version of the esp-idf

thomasfla, 2019-08-21:
A temporary solution is to use this commit of the esp-idf:
git checkout 8d1a9c07a0ff90acc3bad7143ce5154774ec1643

We need to adapt to the new HAL of the ESP32...

thomasfla, 2020-03-30:
@maximekli and @AlexisPotier are working on this

thomasfla, 2020-04-17:
pending #63 addresses this this issue

jviereck, 2020-11-09:
As #63 got merged, is this still required? I tried to compile with the latest 4.1 release of esp-idf but it failed.

thomasfla, 2020-11-09:
@maximekli did update the code for esp-idf v4 and above, but we measured a slightly degraded communication performances, so we decided to keep the v3 for now..

thomasfla, 2022-04-08:
Closing this issue for now, as we won't fix before we prove new espidf performs as well as what we have now. (maybe v5?)

closed 2022-04-08 by thomasfla