open-dynamic-robot-initiative/master-board #10: [SDK] Number of motor boards should not be hardcoded

jviereck, 2019-11-19:
At this point, the number of motor boards is defined at compile time as `N_SLAVES_CONTROLED`. This variable is used over the entire project.

Because this value is hardcoded, it's not possible to support to master boards with different number of controlled boards. For instance, one master board might control a quadruped (using 6 motor boards) and the other master board controlling an arm with only 2 motor boards.

One way to fix this would be to use a second argument to `MasterBoardInterface` like `robot_if(argv[1], num_motor_boards)`.

thomasfla, 2019-11-19:
Actually N_SLAVES_CONTROLED is only used in the example. (The define should
be moved I agree)

On the opposite N_SLAVES is a hard coded value on witch the protocol
depends. ie the firmware is also having this value and expect fix size
packets. Right now, the network protocol is transmitting 6uDrivers data,
even if we only control 2of them.

If needed we could change the protocol and make it adapting to the robot
structure, but this impact the way network packets are handled. I propose
to stick to a fix value for now.

jviereck, 2019-11-19:
Thanks for the explanation! That makes sense then, closing this issue.

closed 2019-11-19 by jviereck