open-dynamic-robot-initiative/master-board #12: [SDK] Use rad and rad/s for motor.position and motor.velocity

jviereck, 2019-11-20:
At this point the `motor.position` and `motor.velocity` returns the quantities as they are reported by the udriver boards. The udriver/ti boards use the units `position: [rotations]` and `velocity: [kilo rotations / minute]`.

I strongly vote we change the values stored and returned by the SDK to SI unites. That is `position: [rad]` and `velocity: [rad/s]`.

thomasfla, 2019-11-20:
This is planned for the refactoring of the API. The only non SI value will be the accelerometer with g values instead of m/s2 since it ease the the gravity compensation that is estimated on-board in the IMU. What do you think?

jviereck, 2019-11-21:
I have a strong opinion to have everything in SI units. Not every user of the sdk might be doing gravity compensation / use the setup for a robotics experiment. Therefore, I would like the used units to be "neutral" / not application specific. This boils down to using SI units.

PR #14 (2019-11-28): Changes motor position, velocities, kp and kd to SI units
This fixes #12 and updates the example to use the new units.
closed 2019-11-28 by paLeziart