open-dynamic-robot-initiative/master-board #18: [SDK] Ability to run as non-root user

jviereck, 2019-11-29:
Right now running the `example.cpp` requires root access on Linux. It would be great to have a way to run the demo without need for root permissions.

nim65s, 2020-01-21:
For the C++ example:
```
sudo setcap cap_net_admin,cap_net_raw+ep master-board-example
./master-board-example eth0
```

For the python example, it is more tricky.
Setting `setcap` on the python interpreter would work, but it's not a great idea.

One solution would be to make a C++ daemon, and use Python with RPC to it instead of bindings.

nim65s, 2020-01-21:
I'll try with a kernel module tomorrow :)

thomasfla, 2020-03-25:
The setcap solution seems good and flexible enough. @AlexisPotier could you add this in the documentation ?

closed 2020-03-27 by thomasfla