open-dynamic-robot-initiative/master-board #19: [SDK] Option to have no printf when running on realtime loop

jviereck, 2019-11-29:
When running on a hard realtime thread, using `printf` can be problematic. At this point, the code is printing a message of ill-sized package length from

```
void MasterBoardInterface::callback(uint8_t src_mac[6], uint8_t *data, int len)
{
  if (len != sizeof(sensor_packet_t))
  {
    printf("received a %d long packet\n", len);
    return;
  }
```

For the hard realtime scenario, it would be good to have a way to log these incidents but not print them.

thomasfla, 2019-11-29:
agree, this was for debug.. We can remove it

paLeziart, 2019-11-29:
I created a pull request for this issue, see #20

paLeziart, 2019-12-05:
The pull request has been merged. I think we can close the issue.

closed 2019-12-06 by thomasfla