open-dynamic-robot-initiative/master-board #21: [SDK] Add helper for calibration of the motors

thomasfla, 2019-11-29:
Motors encoders are not absolute, but incremental + index.
The consequence is that at startup, user have no way to get the absolute position of the joints.
an active calibration procedure is needed to find the INDEX of the encoder.

With the choices that have been made, the calibration could be done with a state machine:

1 Wait for alignment to finish (motor ready?)
2 Do position control to find an index (given a direction of search from the user, or going half a turn in a direction, half a turn in the other, or something else...)
Watch the index detected flag, and go to 3 when detected.
3 Disable position control, and set current to 0A. Enable the index compensation in the uDriver, wait for it to be enabled.
4 The motor is now referenced with respect to the index from the encoder.
5 Optionally, user could give an additional offset to represent a physical offset between the index position and the zero of the robot joint.

All this machinery could be coded in the SDK itself, but to stay in the spirit of the sdk, the calibration function should not be blocking the thread. It is still up to the user to call the send commands and parse packet:

The user will have to write: a loop that is calling:
```
robot_if.ParseSensorData();
robot_if.DoEncoderCalibration();
robot_if.SendCommand();
```

@MaximilienNaveau @jviereck @paLeziart what do you think?

jviereck, 2019-11-30:
+\cc @luator as he did a lot of the calibration work.

Also note the current calibration code is here: https://github.com/open-dynamic-robot-initiative/blmc_robots/blob/master/src/blmc_joint_module.cpp

luator, 2019-12-02:
We are using the `init_homing`/`update_homing` methods of `BlmcJointModule` (linked above by Julian) which do basically exactly what you proposed: Move slowly with a position profile in a user-defined direction until the index is found, then set the zero position as "index position + offset" (this corresponds to homing methods 33/34 in the CiA402 standard).

Since this issue is in the master-board repo, do we need some additional implementation here or could we just use the one from `BlmcJointModule`?  Note, however, that the current implementation there is not fully compatible with the master board as, to my knowledge, the index tick is reported in a different way there.

thomasfla, 2019-12-04:
Thanks for the reference,
Ok, so if we use BLMC robot, and once made compatible, we shouldn't need this homing inside the SDK. 
I propose to still implement a simple homing helper function for the projects that might not depend on BlmcRobot, but only rely on the raw SDK.

jviereck, 2019-12-04:
I strongly second that the sdk should not depend on the blmc_robots package and we should port the code for homing from blmc_robots to the sdk directly.  

That is as blmc_robots introduces some more concrete (like everything is async, everything is build around Timeseries etc). For blmc_robots these design decisions make sense. I doubt these higher constructs are needed for the low-level SDK though and thereby only introduce unnecessary weight.

thomasfla, 2020-03-25:
An example of calibration is available here: https://github.com/thomasfla/solopython/blob/master/masterboard_utils.py

I think calibration should be separated from the SDK, since implementation will largely depend on the robot type: joint calibration order, direction of motion, motion profile, safety limits etc..

The SDK was design not to be a controller, but just provide a raw access to sensor and commands.
A calibration routine is a controller.

closed 2020-03-27 by thomasfla