open-dynamic-robot-initiative/master-board #22: [SDK] Shutdown the robot_if if master-board is not responding

jviereck, 2019-11-29:
Right now it's possible to start a control program, power down the robot, power on the robot again and the same control program takes over the robot again. This is dangerous as the user might not expect the controller to run on the controller right away.

As a safety, the sdk should check if the master-board is responsive or not. If a timeout is reached and the master-board is not responding, the robot_if should shutdown (with possibility to re-enable by the user if desired but needs to be an explicit behavior).

thomasfla, 2019-11-29:
If we want to have the same behavior that we have on the master board, I propose to initialise the timeout only after the first packet as been received, so that we have time to power up the robot. The two system will mutually wait for each-other.

thomasfla, 2020-01-14:
#30 Solves this.

closed 2020-01-14 by thomasfla