open-dynamic-robot-initiative/master-board #24: [sdk] several instances of controller can be launched

thomasfla, 2019-12-06:
As the moment, nothing prevent a user to launch several instances of a controller. This could cause hardware issue if the control jumps back and forth between the two controllers.

I'm not sure how to handle this properly.

Check for control packets on the network socket?
A lock file ?

paLeziart, 2019-12-19:
I guess a lock file could work. A simple .txt which contains 0 by default. When a controller is launched, the first thing we could do is to check the content of this file. If it is 0 then the program resumes and sets the content to 1. The content is set back to 0 when the destructor of MasterBoardInterface is called. 

I don't know if it is enough if we launch two control programs at the same time. I wonder if the two controllers can read a 0 if they are launched really close to each other and the first one does not have the time to set the content to 1 before the second one reads the content of the lock file.

jviereck, 2019-12-19:
Note that there might be cases where you want to run two controllers - that is, when you control two robots from the same computer (think of two solos doing some work together). I think this usecase should be supported.

The way this could be done is by having a lockfile per network interface (or target robot device in case you can communicate to multiple wifi-connected masterboards though the same wifi interface on the computer). 

On unix systems, the easiest way to ensure single access to a resource is via using lockfiles and then remove the lockfile again, once the resource is freed. This means the program / code we run should terminate properly (e.g. when you hit ctrl-c), such that the cleanup / removing the lock procedure is executed.

For implementation details, see [1] about `flock()` and [2].

@MaximilienNaveau , @vincentberenz , do you know if the real_time_tools library has support for filelocks? Would that be a general enough things to add in our utility libraries?

[1]: https://stackoverflow.com/questions/1599459/optimal-lock-file-method
[2]: https://linux.die.net/man/1/lockfile

MaximilienNaveau, 2020-02-06:
I would then use some kind of shared memory mutex if timing is an issue.

thomasfla, 2020-02-12:
I think we could address this issue together with #43 by enhancing the protocol. The Master board would only accept packets from a newly started controller, restarting the timeout in the same time.

A given controller could be identified by a session ID, like a random number, or the current os time..
A newly started controller could also check the active session ID before controlling the robot. 

I won't be able to work on this before next month, but I'm in favor of a solution that is not OS dependent, and also address #43

thomasfla, 2020-03-30:
@maximekli is working on this

thomasfla, 2020-04-17:
pending #63 addresses this this issue

closed 2021-06-02 by thomasfla