open-dynamic-robot-initiative/master-board #41: Using non-realtime thread for reading received messages

jviereck, 2020-01-24:
The core while loop receiving new messages from an interface is defined here:

  https://github.com/open-dynamic-robot-initiative/master-board/blob/master/sdk/master_board_sdk/src/Link_manager.cpp#L70

This runs in a thread, which is non-realtime.

The suggestion is to spawn this thread as an rt-preempt realtime thread. We have a nice abstraction for launching realtime threads in our [real-time-tools package](https://github.com/machines-in-motion/real_time_tools/): If a thread is launched on a non-realtime / non-rtpreempt linux kernel, then a "normal" non-realtime thread is created.

\cc @MaximilienNaveau @thomasfla

thomasfla, 2020-01-27:
I think the thread run with the same priority and scheduling as the parent process and that the behavior is already what we need. 

On my real time system `htop` gives :
[image]
if I run the example with:
`sudo chrt 99 ./master-board-example enp4s2f1`

thomasfla, 2020-01-28:
I will close this issue, feel free to reopen if needed.

closed 2020-01-28 by thomasfla