open-dynamic-robot-initiative/master-board #43: Reconnecting to master_board

jviereck, 2020-01-25:
When the master_board stays powered on, it's not possible to reconnect to it from the PC. Is this supposed to be the case?

For example, in our experimental setup we have:
1) power on the robot with master_board
2) start connection from PC
3) shut down connection from PC
4) start connection from PC again << unable to control the robot

@thomasfla , any idea if this should work? I wonder if there is a timeout on the master board that shuts it down once it detects the communication to the PC at 3) is closed as well.

thomasfla, 2020-01-27:
There is indeed a timeout on the master board for safety.
At the moment, there is two option to reset it: 
  * Restart the master board (witch often mean restarting the whole robot, so the calibration of the encoders is lost)
  * Press the button on the master board, this one will be instantaneous and keeps the calibration of the encoders.

To be able to restart the MB timeout from remote, we need to change the protocol. 
I propose to add a _Session ID_ to the command packet. The timeout would be reset if the session ID has been changed.

jviereck, 2020-01-27:
> Press the button on the master board, this one will be instantaneous and keeps the calibration of the encoders.

In our solo12 setup, the masterboard is not accessable. Is it possible to add a connector / solder to the button to have an external button to reset the board? If not possible, maybe this could be included in a new revision of the masterboard?

> To be able to restart the MB timeout from remote, we need to change the protocol.
I propose to add a Session ID to the command packet. The timeout would be reset if the session ID has been changed.

This sounds good!

thomasfla, 2020-01-27:
Yes, It is possible to connect a push button between the port SW0 and GND on the masterboard V2.

thomasfla, 2020-03-30:
@maximekli is working on this

thomasfla, 2020-04-17:
pending #63 addresses this this issue