open-dynamic-robot-initiative/master-board #45: [SDK] Installation of Python3 sdk library goes in python3/dist-packages instead of python3.5/dist-packages

thomasfla, 2020-01-29:
When installing the SDK library for python3, the .so goes in `/usr/local/lib/python3/dist-packages` on Ubuntu. This is not the standard, and python will not find the library by default.

A temporary fix is:
`sudo sh -c "echo /usr/local/lib/python3/dist-packages >> /usr/local/lib/python3.5/dist-packages/local.pth"`

thomasfla, 2020-02-14:
@nim65s Do you have a solution on this issue?

nim65s, 2020-02-14:
I tried many things, and this seems to work:

```
FROM ubuntu:16.04

WORKDIR /root
RUN apt update -qqy \
 && apt upgrade -qqy \
 && apt install -qqy \
    git \
    cmake \
    g++ \
    make \
    python3 \
    libpython3-dev \
    libboost-all-dev \
    doxygen
RUN git clone --recursive https://github.com/open-dynamic-robot-initiative/master-board.git
RUN mkdir /root/master-board/sdk/master_board_sdk/build
WORKDIR /root/master-board/sdk/master_board_sdk/build
RUN cmake \
    -DBUILD_PYTHON_INTERFACE=ON \
    -DPYTHON_EXECUTABLE=/usr/bin/python3 \
    -DPYTHON_STANDARD_LAYOUT=ON \
    -DPYTHON_DEB_LAYOUT=ON \
    .. \
 && make \
 && make install
WORKDIR /root
ENV LD_LIBRARY_PATH /usr/local/lib
RUN python3 -c 'import libmaster_board_sdk_pywrap; print(libmaster_board_sdk_pywrap.__file__)'
```

The point is to set both `-DPYTHON_STANDARD_LAYOUT=ON -DPYTHON_DEB_LAYOUT=ON`, but this feel really hacky.

ref https://github.com/jrl-umi3218/jrl-cmakemodules/pull/234 + replace `site` by `dist` for Debian.

closed 2020-11-13 by jviereck

RX-00, 2024-06-10:
Hi, I was able to compile and install the sdk on Ubuntu 22.04 but ran into this undefined symbol error

`
ImportError: /usr/local/lib/python3.10/dist-packages/libmaster_board_sdk_pywrap.cpython-310-x86_64-linux-gnu.so: undefined symbol: _ZNK5boost6python7objects21py_function_impl_base9max_arityEv
`

I noticed that @olivier-stasse  had a similar issue on this [resolved issue](https://github.com/open-dynamic-robot-initiative/master-board/pull/160). However, I am not clear on how updating cmake fixes the issue, which it doesn't for me. Currently my cmake version is 3.22.1. Is there a specific version that I'm missing?

olivier-stasse, 2024-06-11:
Dear @RX-00,
 my message mentionned the `cmake module`  which does not relate to cmake as the framework from Kitware (https://cmake.org/) but rather the git submodule named `cmake`  in https://github.com/open-dynamic-robot-initiative/master-board/tree/master/sdk/master_board_sdk. Here you will see that the directory cmake point towards the github repository jrl-cmakemodules  (https://github.com/jrl-umi3218/jrl-cmakemodules/tree/5275edfd1a8c6d45e0f90177edad93ef15b2008f). This set of cmake scripts is helping us packaging our software on various platforms. It was this git submodule which was not up to date. To do so you can do:

olivier-stasse, 2024-06-11:
Finally, I would advise next time to comment the resolved issue you mentionned rather than this one which is not related.

RX-00, 2024-06-11:
Thank you! Apologies for commenting on the wrong resolved issue. My thinking was that since this thread was about a Linux machine (whereas the other was titled about Mac) and the fact that I ran into this error by going through the steps listed above.

commit 95481c1e (2020-11-13): CMake: fix sdk installation path, fix #45