open-dynamic-robot-initiative/master-board #60: Document connection between master-board and IMU

jviereck, 2020-03-30:
For the connection between the master-board and the IMU a custom wire is used. So far we are missing documentation on how the wire is produced / the pins are connected and which connectors are used (one end is a 7-pin female Hirose connector, but I am not sure about the other side).

thomasfla, 2020-03-30:
Thanks for pointing this out @jviereck 
I've just add some doc here: https://github.com/open-dynamic-robot-initiative/master-board/blob/master/documentation/masterboard_wiring.md

I only documented the V2 wiring with the dedicated 4pin Hiroses connector. The V1 is to be considered obsolete for interfacing with IMU, since the voltage regulator on MB is not powerful enough to power external sensors.

closed 2020-03-30 by thomasfla

jviereck, 2020-03-30:
Awesome, thanks so much for the documentation! Not supporting V1 anymore makes sense.