open-dynamic-robot-initiative/master-board #68: Problem of connecting to LAN interface

MiladShafiee, 2020-08-24:
Hello everyone,

Thanks for providing this usefull SDK. I'm using Ubuntu 18.04 that has been connected to a network for internet connection. I had a problem of Ethernet connection that was showing "connecting" and after a while a failing connection error.

I solved this problem by setting the IP address and Netmask in the Ethernet connection setting and I add some screenshot of my settings, it may be useful for other users if they face the similar issue::

[image]

[image]

[image]

PRGeeesus, 2021-01-21:
Hi, i am facing a similar Issue: 

When I try to load the example.pyc with the method as mentioned in:
https://github.com/open-dynamic-robot-initiative/master-board/blob/master/sdk/master_board_sdk/README.md

i get:
[image]

I was not able to fix this using the method mentioned by @MiladShafiee.
when booting, my master board is in the WIFI_ETH_ETH_ERROR state.
I really can not find a way to properly debug this. I also dont have any way of monitoring the packages sent.
I have traced this back to the "master_board_interface.cpp" file, where in the Init() signal(SIGINT, KeyboardStop) is called.

jviereck, 2021-01-21:
Hi @PRGeeesus , 

a few questions.

can you describe your setup a bit? You managed to flash the master board and now you try to connect via LAN? Are you sure the enp3s0 is the correct interface name?

Did you try to run the cpp examples?

Best,
Julian

PRGeeesus, 2021-01-22:
Hi Julian.
I am on Ubuntu 20.04 LTS and managed to flash the master-board firmware. I am now trying to connect it via LAN. (As i dont have any Wifi that supports monitor or injection mode.)
I know that enp3s0 is the correct interface as my Internet is gone when i try to set up the flashing Via LAN. Also "ifconfig" shows only this interface next to lo (which i assume is localhost)
The master-board is powered via a power supply. I assume that the firmware is flashed correctly as the LED is changing colours (from which I assumed the states are entered as described in (https://github.com/open-dynamic-robot-initiative/master-board/blob/master/documentation/masterboard_state_machine.md)). The 

When I power up the board, some colour lights up (can't really tell as it is too fast). After that it immediatly switches to the Red blinking.
When i try to run the .cpp example make runs without any issue.
The "sudo ./bin/exec enp3s0"
returns
"sudo: ./bin/exec: command not found"
from which i assume it is meant to say "sudo ./bin/example" ?
That gives me:
[image]

thank you very much for your help,

Martin

jviereck, 2021-01-23:
Hi Martin,

not sure what is going in there. Can you install wireshark [1] and check if you see any packages being sent to the master board and back?

Best,
Julian

[1]: https://www.wireshark.org/

PRGeeesus, 2021-01-23:
I checked with an oscilloscope and no Data is transmitted from the PC to the master board. I am not confident enough with Ubuntu to know what my next steps would be to debug this. My next step would be to try it on Monday on another machine :/

jviereck, 2021-01-24:
Hi Martin,

there should be at least some packages coming from the PC towards the master board. This indicates the setup from the PC is not properly I guess.

Did you follow the steps Milad outlined in the first post?

Best,
Julian