open-dynamic-robot-initiative/master-board #73: The velocity resolution is poorly chosen

thomasfla, 2020-10-28:
I realize that we might have poorly chosen the velocity resolution in the communication protocol.

Indeed, as documented here https://github.com/open-dynamic-robot-initiative/master-board/blob/master/documentation/BLMC_%C2%B5Driver_SPI_interface.md#data-representation
the velocity have a resolution of 2^-11 krpm witch correspond to a motor velocity 0,48 rad/s.
With a gearbox of 9:1 it is 0,00568 rad/s at the joint level.
The velocity is encoded with 16bits and the full scale is about -16000rpm to 16000rpm. 

We should agree on a lower full scale in order to increase the resolution according to our application.

Once we agree on a new fixed point value for this data representation, the next issue is that we need to upgrade all our robot and drivers at the same time.
Indeed if the driver and the robot have different data representation, we will have highly overestimate velocities measurements and this can be dangerous for our controller.

I do not have yet a good solution on how to proceed, but ideally this update would be incompatible with the current firmwares so we can not miss interpret the data.

A mechanism like this is already implemented in the PC to master board protocol (it includes and check a protocol version)
but nothing is implemented for the masterboard to uDrivers com. We could maybe change the CRC seed ? It's important that we keep this messages as short as possible, and with the CRC solution we keep the same message length.

The CRC is not transmitted to the PC, so we will need to:

- Update the udrivers with new velocity range and a new CRC seed.
- Update the Master board with a new protocol version and the new CRC seed.
- Update the sdk with new protocol version and the new velocity range.

What do you think ? 
From users, what is the optimal range for the velocity?

jviereck, 2020-11-02:
I feel a resolution of `0,00568 rad/s at the joint level.` is good enough for practical usage. I wonder therefore if changing the resolution makes sense.

If that helps to choose a new range of values: For the jumping task on solo12 I never exceeded 100 rad/s as joint level velocity, meaning 900 rad/s at the motor level. We should add some room for faster jumps so maybe say 500 rad/s at the joint level should be enough?

One thing we could do: Reduce the number of bits used for the velocity and use the free bits for an improved/more states on the error code?

thomasfla, 2020-11-02:
Ok, so if there is no need for a better velocity resolution, let's keep the actual choice.

If we want to augment the error codes we should instead use the 3 free bits in the status field: https://github.com/open-dynamic-robot-initiative/master-board/blob/master/documentation/BLMC_%C2%B5Driver_SPI_interface.md#status
Those free bits were actually put here inorder to be able to augment the error size. With a total of 7bits we can encode 128 different error.

MaximilienNaveau, 2021-06-09:
@thomasfla  and @jviereck, It seems I can close this

closed 2021-06-09 by MaximilienNaveau