open-dynamic-robot-initiative/master-board #79: adding the circuit for entering in flash mode directly on master board

ferdinand-wood, 2020-12-22:
As described in "[Flashing the firmware](https://github.com/open-dynamic-robot-initiative/master-board/blob/master/firmware/README.md#flashing-the-firmware)", the esp chip needs the RST and GPIO-0 pins lowered to enter the flash mode. On the WeMos D1 mini board, this is done by two transistors shown [here](https://escapequotes.net/wp-content/uploads/2016/02/2020-02-10_12-06-36.jpg). This circuit could be directly implemented on a new generation of the solo's master board. With this, it should be possible to program the master board without any additional hardware and just using the DTR and RTS pins from a usb-serial-connector.
I don't know if this is feasible, so feel free to comment the idea.

MaximilienNaveau, 2021-03-04:
Hi,
I am all for the suggestion but I am a noob in electronics, @thomasfla is the expert.

I would suggest you post your suggestion on this discourse forum where we talk about design and software a bit more in detail and more broadly:
https://odri.discourse.group/

If you already did post in the forum please copy the link to your post in this issue as well!
This way we keep track to where the answer to this issue was.

ferdinand-wood, 2021-03-04:
Hey there, I have postet the idea on the forum: https://odri.discourse.group/t/improvement-idea-for-master-circuit-board-add-circuit-on-master-board-to-entering-in-flash-mode/209