open-dynamic-robot-initiative/master-board #104: [SDK] Get Error Code

MusubaPy, 2022-01-02:
Greetings,

I went to this for a long time and finally I was able to completely assemble your project.

I ran into a problem: I sometimes get an error while running _example.cpp._

I cannot understand what the error might be related to. Sometimes an error occurs when torque is applied to the links. Sometimes it occurs when I send a position to a link.

I am trying to use the **GetErrorCode()** function, but I am not getting the desired result.

P.S. I only use half of the drivers and motors.

[image]

thomasfla, 2022-01-03:
Dear Roaman, 

The error codes are declared here https://github.com/open-dynamic-robot-initiative/udriver_firmware/blob/782998348fbe88158f398aadde11c36a4162ba12/firmware/mw_dual_motor_torque_ctrl/src/spiapi.h#L167

The Error 1 and 6 correspond to encoder errors, this can happen if the electrical connection with the encoder is bad, or if the encoder code wheel is damaged/miss aligned. 

Maybe some context on how such error is triggered might help you to solve your issue.
The encoders are incremental, and the impulses are integrated to know the position. In addition to that, the encoders have an index pulse that are fired every motor turns. To check if we loose encoder steps, we check every time the index pulse is seen if the position has varied of exactly one turn. This explains why the error is not triggered instantly, but only after the encoder sees an index pulse, so after manually turning the joint, or commanding a move.

I hope this helps, 

We should better document the error codes and write some troubleshooting section somewhere. I will try to work on this soon.

thomasfla, 2022-04-08:
related to #70

thomasfla, 2022-10-14:
Hi, the encoder is equiped with an index line (impulse at every motor turn) in the firmware of the udriver, we check that at every index interrupt, we have moved from one (or minus one) full turn (with some margin threshold)
If we don't, this means we have lost some encoders ticks, and we don't know where we are. This is error 1.

This can be caused by:
* Dirty/ misaligned code wheel
* Bad connection on the encoder wires / connector
* Extreme electromagnetic perturbations

Hoping this helps..

dvogureckiy99, 2022-10-14:
@thomasfla  Thank you very much, but it error happens after my motors turn around a little. If I set amplitude less  than math.pi/6 then this error doesn't happen. Can be it related to rotation speed ?

dvogureckiy99, 2022-12-22:
@thomasfla Hi, again, I  putted the corresponding Codewheel PWB Mounts Encoder and I have got impulses on the pin I on the udriver, i.e. impulses of the encoder index line (impulse at every motor turn), 
[image]
but I still getting error 6 or 1.

thomasfla, 2022-12-22:
Hi,

This still resemble in a misaligned code week.
You can have a look at the encoder datasheet to get an idea of the mechanical tolerances. 

You can also probe the a, b and I line together with a scope or a logic analyser so help troubleshooting.

Best, 
thomas.

dvogureckiy99, 2022-12-23:
@thomasfla and can I ask you to send a photo from the oscilloscope of what all three signals A,B,I should look like?