open-dynamic-robot-initiative/master-board #106: Raspberry Pi

emrecemal, 2022-01-20:
Hello,

I am trying to run the example motor control code on Raspberry Pi, but I couldn't cmake on it. Do you know how to do that?

thomasfla, 2022-01-20:
Hello, 

Can you tell us what cmake is reporting? 
Maybe @nim65s can help with the compiling for Raspberry Pi.

emrecemal, 2022-01-20:
Yes, initially I realised that the boost library is missing. I installed it using the following codes:

`sudo apt-get install libboost1.74-*`

Then I did the following, and here is the error I get:

`pi@bunny:~/master-board/sdk/master_board_sdk/build $ cmake -DBUILD_PYTHON_INTERFACE=ON -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=$(which python3) ..
-- Package version (ROS package.xml): 1.0.0
-- The CXX compiler identification is GNU 10.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Performing Test R-pedantic
-- Performing Test R-pedantic - Success
-- Performing Test R-Wno-long-long
-- Performing Test R-Wno-long-long - Success
-- Performing Test R-Wall
-- Performing Test R-Wall - Success
-- Performing Test R-Wextra
-- Performing Test R-Wextra - Success
-- Performing Test R-Wcast-align
-- Performing Test R-Wcast-align - Success
-- Performing Test R-Wcast-qual
-- Performing Test R-Wcast-qual - Success
-- Performing Test R-Wformat
-- Performing Test R-Wformat - Success
-- Performing Test R-Wwrite-strings
-- Performing Test R-Wwrite-strings - Success
-- Performing Test R-Wconversion
-- Performing Test R-Wconversion - Success
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
CMake Warning at cmake/doxygen.cmake:496 (MESSAGE):
  Failed to find Doxygen, documentation will not be generated.
Call Stack (most recent call first):
  cmake/post-project.cmake:59 (_SETUP_PROJECT_DOCUMENTATION)
  CMakeLists.txt:32 (PROJECT)

-- currently used  C++ standard: 201402
-- minimal C++ standard upgraded to 11
-- Found Python3: /usr/bin/python3 (found version "3.9.2") found components: Interpreter Development Development.Module Development.Embed 
-- PythonLibraryDirs: /usr/lib/arm-linux-gnueabihf
-- PythonLibVersionString: 3.9.2
-- Python site lib: lib/python3.9/site-packages
CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1311 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1919 (_Boost_MISSING_DEPENDENCIES)
  cmake/boost.cmake:27 (FIND_PACKAGE)
  cmake/boost.cmake:106 (SEARCH_FOR_BOOST_COMPONENT)
  CMakeLists.txt:43 (SEARCH_FOR_BOOST_PYTHON)

CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1311 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1919 (_Boost_MISSING_DEPENDENCIES)
  cmake/boost.cmake:27 (FIND_PACKAGE)
  cmake/boost.cmake:106 (SEARCH_FOR_BOOST_COMPONENT)
  CMakeLists.txt:43 (SEARCH_FOR_BOOST_PYTHON)

CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
[... 32 lines cut ...]
                 from /home/pi/master-board/sdk/master_board_sdk/include/master_board_sdk/Link_manager.h:4,
                 from /home/pi/master-board/sdk/master_board_sdk/include/master_board_sdk/ESPNOW_manager.h:6,
                 from /home/pi/master-board/sdk/master_board_sdk/src/ESPNOW_manager.cpp:7:
/usr/include/c++/10/bits/stringfwd.h:74:11: note: declaration of ‘std::string’ {aka ‘class std::__cxx11::basic_string<char>’}
   74 |     class basic_string;
      |           ^~~~~~~~~~~~
/home/pi/master-board/sdk/master_board_sdk/src/ESPNOW_manager.cpp: In member function ‘void ESPNOW_manager::unset_filter()’:
/home/pi/master-board/sdk/master_board_sdk/src/ESPNOW_manager.cpp:16:3: error: ‘free’ was not declared in this scope
   16 |   free(this->bpf.filter);
      |   ^~~~
/home/pi/master-board/sdk/master_board_sdk/src/ESPNOW_manager.cpp:8:1: note: ‘free’ is defined in header ‘<cstdlib>’; did you forget to ‘#include <cstdlib>’?
    7 | #include "master_board_sdk/ESPNOW_manager.h"
  +++ |+#include <cstdlib>
    8 | 
/home/pi/master-board/sdk/master_board_sdk/src/ESPNOW_manager.cpp: In member function ‘void ESPNOW_manager::set_filter(uint8_t*, uint8_t*)’:
/home/pi/master-board/sdk/master_board_sdk/src/ESPNOW_manager.cpp:94:36: error: ‘malloc’ was not declared in this scope
   94 |  this->bpf.filter = (sock_filter*) malloc(sizeof(sock_filter)*this->bpf.len);
      |                                    ^~~~~~
/home/pi/master-board/sdk/master_board_sdk/src/ESPNOW_manager.cpp:94:36: note: ‘malloc’ is defined in header ‘<cstdlib>’; did you forget to ‘#include <cstdlib>’?
make[2]: *** [CMakeFiles/master_board_sdk.dir/build.make:85: CMakeFiles/master_board_sdk.dir/src/ESPNOW_manager.cpp.o] Error 1
make[2]: Leaving directory '/home/pi/master-board/sdk/master_board_sdk/build'
make[1]: *** [CMakeFiles/Makefile2:203: CMakeFiles/master_board_sdk.dir/all] Error 2
make[1]: Leaving directory '/home/pi/master-board/sdk/master_board_sdk/build'
make: *** [Makefile:163: all] Error 2
`

nim65s, 2022-01-21:
I @emrecemal,

I already fixed that, but I have a hard time merging my recent work on the main branch. I'll cherry-pick this fix and push it to master for you. Sorry for the inconvenience.

commit 8fe4757a (2022-01-21): SDK: cherry-pick fixes, fix #106

initial commit: 32b0252c64a064c55e8cd07115a003b15c0d0a86
closed 2022-01-21 by nim65s