open-dynamic-robot-initiative/master-board #108: [Firmware] communication stops about 17 minutes after startup.

thomasfla, 2022-05-17:
@luator reported that the communication fails around 17m 45s-50s after the boards are powered one.

luator, 2022-05-17:
To add a bit more detail:
- It fails with "error 2" (SPI timeout) on the motor boards.
- Most of the time, board 5 (i.e. the last one in the list) fails first, not sure if this is of any significance.
- There are no bad CRC checksums in the SPI communication before it fails, so it is not caused by broken pakets.

jviereck, 2022-05-17:
Have you tried to run the master board without any motor boards? This would allow to check if the udrivers cause the issue or the master board.

luator, 2022-05-18:
No, I haven't tried that yet.  I'm a bit unsure what to expect in this case, though.  Since it is somehow an issue in the SPI communication, it might be that even if the master board is causing it, the situation will simply not arise if no motor board is connected.  Maybe it would make sense to test with motor boards but powering them separately, so they don't start at the same time (e.g. power the motor boards, wait 5min, then power the master board and see how long it takes until the error).

luator, 2022-10-10:
Looks like the cause of the issue has been found: https://github.com/open-dynamic-robot-initiative/udriver_firmware/issues/13