open-dynamic-robot-initiative/master-board #116: Overflow in fixed point conversion

tlbtlbtlb, 2022-09-16:
When converting floating point host numbers to fixed-point integers for sending to the masterboard, there's no consideration of overflow. This is somewhat difficult to prevent at the higher level, since the numbers are all multiplied by gear ratios and encoder counts before being converted to fixed-point.

For instance, Kp overflows at about 5.1 for the Solo12 robot, but you have to know several other parameters in order to calculate that. The overflow point isn't far from reasonable values people might want to use (I currently use 4.0). Kd overflows at 0.4, when a common value is 0.2.
```c++
command_packet.dual_motor_driver_command_packets[i].kp[0] = FLOAT_TO_D16QN(2. * M_PI * motor_drivers[i].motor1->kp, UD_QN_KP);
...
#define UD_QN_KP   11
...
#define FLOAT_TO_D16QN(a,n)       ((int16_t) ((a) * (1<<(n))))
...
motors_[i]->set_kp(desired_gains[i] / (gear_ratios_[i] * gear_ratios_[i] * motor_constants_[i]));
```

I suggest clamping these values to within the representable range, rather than overflowing to negative numbers. Something like:

```
#define FLOAT_TO_D16QN(a,n)       ((int16_t) (min(max((a) * (1<<(n)), -32765.0), +32765.0))
```

https://github.com/open-dynamic-robot-initiative/master-board/blob/8ecd488332a97818fc1ee963f86d147090f0ec2f/sdk/master_board_sdk/src/master_board_interface.cpp#L198
https://github.com/open-dynamic-robot-initiative/master-board/blob/347dfa697b179550545e7c7696e7666ffe543734/sdk/master_board_sdk/include/master_board_sdk/protocol.h#L137

thomasfla, 2022-09-16:
Hi, thanks for opening this issues, this is indeed something that was on a todo list and got lost a long time ago. We should definitely fix it.

A few remarks:

- We want to cover all cases, for signed and unsigned variables, 8, 16 and 32bits. 

- About this, looking at the code, it seems we have another issue. Kp and Kd are unsigned values (we don't want to have diverging controllers !) and are fortunately treated like so by the motor controller firmware. But in the sdk, I think we should have called the FLOAT_TO_uD16QN marco instead of FLOAT_TO_D8QN for these values. In practice, this error might have cut in half the range value for the unsigned values (not sure, should check). If so, fixing this will help with the tight margins you pointed out.

- We might want to decide what to do in case of overflow. Indeed, for example when tuning, it might be misleading to think that the gain is set to some values when instead it has been clamp. We could generate an error or a warning, stop the program or continue with a clamped value. Any feedback is welcome on this. I personally like that a very high value generate an emergency stop instead of clamping. Anyway, It would be nice to expose the maximum ranges to the user with some function.

tlbtlbtlb, 2022-09-17:
As a user, I vote for:

- `kp`, `kd`, `i_sat`, `current_ref`, `velocity_ref`: saturate, maybe print a warning the first time it happens
- `position_ref`: set a flag with similar behavior to `MasterBoardInterface::timeout`, since it must be a nonsense value.

`kp` and `kd` must indeed be `uint16_t` inside the motor controller, because the robot didn't go bananas when I put in a too-large value that wrapped around to negative. It just gave a low stiffness.

And yes, it would be nice if `Motor` had functions like `GetKpLimits` which you might call prior to calling `SetKp`. And then `odri_control_interface::JointModules` could have functions to get the limits scaled by gear ratios. If you decide to take more drastic action on overflow, these functions become essential.

https://github.com/open-dynamic-robot-initiative/master-board/compare/master...tlbtlbtlb:master-board:master

thomasfla, 2022-09-19:
Thanks, this looks good to me. 
If you have time, could you also fix the FLOAT_TO_uD16QN instead of FLOAT_TO_D16QN for Kp and Kd and open a PR once tested? 

Thanks for your contribution !

tlbtlbtlb, 2022-09-20:
I submitted PR #117 with my proposed fixes. I took the liberty of adding unit tests using the catch2 test system. `make test` in sdk/master_board_sdk will run them, if you have catch2 installed. See https://github.com/catchorg/Catch2 if you don't already have it. It doesn't depend on catch2 unless you explicitly run `make test`.

nim65s, 2022-09-28:
#117 is merged :)

But on my system, I don't have the same behavior for `quiet_NaN`:
```
../tests/test_protocol.cpp:27: FAILED:
  CHECK( ((int32_t) std::min(std::max(((std::numeric_limits<double>::quiet_NaN()) * (1<<(24))), -2147483648.0), 2147483648.0)) == int32_t(0) )
with expansion:
  -2147483648 == 0
  ```
  
  What should we do about it ?

olivier-stasse, 2024-06-21:
Dear all, I propose to close this issue as it was not active for a long time.
Feel free to reopen it if needed.

closed 2024-06-21 by olivier-stasse