open-dynamic-robot-initiative/master-board #124: CMake Error at CMakeLists.txt:## (INCLUDE)

dvogureckiy99, 2023-02-10:
When I tryed command `make -DBUILD_PYTHON_INTERFACE=ON -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=$(which python3) .. ` I have got error:
```
CMake Error at CMakeLists.txt:26 (INCLUDE):
  INCLUDE could not find load file:

    cmake/base.cmake

CMake Error at CMakeLists.txt:27 (INCLUDE):
  INCLUDE could not find load file:

    cmake/boost.cmake

CMake Error at CMakeLists.txt:28 (INCLUDE):
  INCLUDE could not find load file:

    cmake/python.cmake

CMake Error at CMakeLists.txt:29 (INCLUDE):
  INCLUDE could not find load file:

    cmake/ide.cmake

CMake Error at CMakeLists.txt:31 (COMPUTE_PROJECT_ARGS):
  Unknown CMake command "COMPUTE_PROJECT_ARGS".

-- Configuring incomplete, errors occurred!
```

dvogureckiy99, 2023-02-10:
I was wrong.

nim65s, 2023-02-13:
you need to either clone this repository with `git clone --recursive` to get the cmake submodule, or call `git submodule update --init` afterwards`

closed 2023-02-13 by nim65s