open-dynamic-robot-initiative/master-board #125: Timeout while waiting for ack

dvogureckiy99, 2023-02-10:
When I run example `sudo PYTHONPATH=. python3 example/example.pyc -i eth0`, I have got this:
```
-- Start of example script --
if_name: eth0
Using Ethernet (eth0)
last = 5403.956032924
Shutting down connection (eth0)
Timeout while waiting for ack.
Shutting down connection (eth0)
Masterboard timeout detected.
Either the masterboard has been shut down or there has been a connection issue with the cable/wifi.
-- End of example script --
```

dvogureckiy99, 2023-02-10:
solved

omertarbanus, 2023-10-12:
Hi, how did you solve this? I am having the same issue

dvogureckiy99, 2023-10-23:
one of the reason that can occur there is you should power on master-board first then udriver.
or your power on master-board isn't. 
or ethernet name is wrong

tbanus, 2023-10-24:
Hi,  in my case the masterboard came with an old driver installed from pal robotics. when i tried to run it with the latest code, pc and board did not understand each other.

rolling back to an older commit from april (when pal robotics assumably produced the robot) fixed the problem.

havent tried updating the masterboard since esp packages are not very current and it is too much trouble to find old versions

dvogureckiy99, 2023-10-24:
what do you mean by `old driver` ?

dvogureckiy99, 2023-10-24:
Honestly I again encountered with this problem and can't solve this.

dvogureckiy99, 2023-10-24:
I download driver for my ethernet interface `RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller`:
https://www.realtek.com/en/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-pci-express-software

dvogureckiy99, 2023-10-24:
Trying change name of ethernet, cause my previous `logical name` was `enp1s0`. But as I see in your program code:
https://github.com/open-dynamic-robot-initiative/master-board/blob/d8e07a7e4dea1214706395c0c3591469dc403a1b/sdk/master_board_sdk/src/master_board_interface.cpp#L59C14-L59C14

dvogureckiy99, 2023-10-24:
> Hi, in my case the masterboard came with an old driver installed from pal robotics. when i tried to run it with the latest code, pc and board did not understand each other.
>
> rolling back to an older [...]

please, tell me exactly what commit ID you use ?

MaximilienNaveau, 2023-10-24:
Basically you need to always have the version of  the drivers and the firmware which are compatible.
I suggest you spend some time reading on how to flash the master board.
It's a little annoying at the beginning but then it will well when updates will come.

If I recall the documentation of how to are in the readme of the repository in the firmware folder:
https://github.com/open-dynamic-robot-initiative/master-board/blob/master/firmware/README.md

dvogureckiy99, 2023-10-24:
Thanks, but it will be better if you somehow specify the version of firmware and driver. Cause already flash master-board before using new driver is little annoying.

MaximilienNaveau, 2023-10-24:
Well either you flash your own or you ask you supplier which version was flashed.
I strongly suggest to setup a computer with the software to flash the board and keep it next to the robot. It's a time well spent trust me😉
Plus when you flash a board write down either the tag or the commit number.

When you download this repo each tag correspond to a compatible firmware vs Driver Software.

dvogureckiy99, 2023-10-24:
Thank you very much, I found version with which I flash master-board: commit d712d55

scakki, 2024-04-11:
Hello,

I am facing the same issue with running the example.py file as shown below. I believe I have flashed the master board using the 8d1a9c0 commit as suggested in the GitHub README. Are there any other possible issues that I need to look into? 
Thank you.

[image]

make flash output-
> Generating esp32.project.ld
> LD build/masterboard.elf
> esptool.py v2.7-dev
> Flashing binaries to serial port /dev/ttyUSB0 (app at offset 0x10000)...
> esptool.py v2.7-dev
> Serial port /dev/ttyUSB0
> Connecting....
> Chip is ESP32D0WDQ5 (revision 1)
> Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
> MAC: 0c:8b:95:c6:97:3c
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 2000000
> Changed.
> Configuring flash size...
> Auto-detected Flash size: 4MB
> Flash params set to 0x0220
> Compressed 24736 bytes to 14815...
> 
> Writing at 0x00001000... (100 %)
> Wrote 24736 bytes (14815 compressed) at 0x00001000 in 0.1 seconds (effective 1874.2 kbit/s)...
> Hash of data verified.
> Compressed 649760 bytes to 400323...
> 
> Writing at 0x00010000... (4 %)
> Writing at 0x00014000... (8 %)
> Writing at 0x00018000... (12 %)
> Writing at 0x0001c000... (16 %)
> Writing at 0x00020000... (20 %)
> Writing at 0x00024000... (24 %)
> Writing at 0x00028000... (28 %)
> Writing at 0x0002c000... (32 %)
> Writing at 0x00030000... (36 %)
> Writing at 0x00034000... (40 %)
> Writing at 0x00038000... (44 %)
> Writing at 0x0003c000... (48 %)
> Writing at 0x00040000... (52 %)
> Writing at 0x00044000... (56 %)
> Writing at 0x00048000... (60 %)
> Writing at 0x0004c000... (64 %)
> Writing at 0x00050000... (68 %)
> Writing at 0x00054000... (72 %)
> Writing at 0x00058000... (76 %)
> Writing at 0x0005c000... (80 %)
> Writing at 0x00060000... (84 %)
> Writing at 0x00064000... (88 %)
> Writing at 0x00068000... (92 %)
> Writing at 0x0006c000... (96 %)
> Writing at 0x00070000... (100 %)
> Wrote 649760 bytes (400323 compressed) at 0x00010000 in 6.1 seconds (effective 857.9 kbit/s)...
> Hash of data verified.
> Compressed 3072 bytes to 103...
> 
> Writing at 0x00008000... (100 %)
> Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 6836.9 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Hard resetting via RTS pin...

make menuconfig output-
[... 49 lines cut ...]
> Starting scheduler on PRO CPU.[0m
> [0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
> ETH/WIFI init size 4
> ETH/WIFI command size 172
> ETH/WIFI ack size 5
> ETH/WIFI sensor size 208
> SPI size 34
> [0;32mI (35) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
> [0;32mI (35) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
> [0;32mI (35) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
> [0;32mI (35) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
> [0;32mI (35) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
> [0;31mE (1035) emac: Timed out waiting for PHY register 0x2 to have value 0x0007(mask 0xffff). Current value 0xffff[0m
> [0;31mE (2035) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0(mask 0xfff0). Current value 0xffff[0m
> [0;31mE (2035) emac: Initialise PHY device Timeout[0m
> ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400878b8
> file: "/home/sakki/devel/master-board/firmware/components/direct_ethernet/direct_ethernet.c" line 197
> func: eth_init
> expression: esp_eth_enable()
> 
> ELF file SHA256: 6c74a5518366d2146adc49368c4101bbcc123c0ad4f3fe46c029a64ea7e429c1
> 
> Backtrace: 0x40087444:0x3ffbb530 0x400878bb:0x3ffbb550 0x400e7545:0x3ffbb570 0x400d3a8d:0x3ffbb5e0 0x400d136e:0x3ffbb620 0x4008c701:0x3ffbb640
> 
> Rebooting...

thomasfla, 2024-04-12:
Hi, did you run this command successfully? ` espefuse.py set_flash_voltage 3.3V` (from https://github.com/open-dynamic-robot-initiative/master-board/blob/master/firmware/README.md )

scakki, 2024-04-12:
Yes. I was able to run it successfully.

thomasfla, 2024-04-12:
For some reason, it seems that you are not able to enable the Ethernet on your masterboard. This is a hardware issue, I think. 
You can take a look at the soldering around the Ethernet PHY (LAN8720, U$2).

scakki, 2024-04-12:
Thank you for your response.
I am currently looking into the Ethernet port and trying to verify with a different Master Board. Additionally, while running make menuconfig we get the following error. Does that indicate that the flashing is not yet completed? The master board still has the solid white light on, which indicates setup mode as per the [GitHub documentation](https://github.com/open-dynamic-robot-initiative/master-board/blob/master/documentation/masterboard_state_machine.md).

> Toolchain path: /home/sakki/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
> WARNING: Toolchain version is not supported: esp-2019r2
> Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> WARNING: Compiler version is not supported: 8.2.0
> Expected to see version(s): 5.2.0
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> /home/sakki/esp/esp-idf/tools/check_python_dependencies.py:22: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
>   import pkg_resources
> Python requirements from /home/sakki/esp/esp-idf/requirements.txt are satisfied.
> make[1]: Entering directory '/home/sakki/esp/esp-idf/tools/kconfig'
> cc -c -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MMD  /home/sakki/esp/esp-idf/tools/kconfig/mconf.c -o mconf.o
> as: unrecognized option '--64'
> make[1]: *** [Makefile:171: mconf.o] Error 1
> make[1]: Leaving directory '/home/sakki/esp/esp-idf/tools/kconfig'
> make: *** No rule to make target '/home/sakki/esp/esp-idf/tools/kconfig/conf-idf', needed by '/home/sakki/esp/esp-idf/tools/kconfig/mconf-idf'.  Stop.