open-dynamic-robot-initiative/master-board #126: [doc] Error code

dvogureckiy99, 2023-02-10:
What this error mean, I connected one driver board to the master-board and after startup `sudo PYTHONPATH=. python3 example/listener.pyc -i enp1s0`. Why it doesn't show anything ?:
[image]
When I run `sudo PYTHONPATH=. python3 example/example.pyc -i enp1s0`:
```
    |     accelerometer    |       gyroscope      |       attitude       |  linear acceleration |
IMU |   0.00   0.00   0.00 |   0.00   0.00   0.00 |   0.00   0.00   0.00 |   0.00   0.00   0.00 |

ADC 04 |  0.440 |  0.582 |

Motor | enabled | ready | IDXT | Index det |    position   |    velocity   |    current    |
   08 |       1 |     1 |    0 |         1 | -6.761350e-03 |  0.000000e+00 |  4.101562e-02 | 
   09 |       1 |     1 |    0 |         0 | -6.761350e-03 |  0.000000e+00 |  1.562500e-02 | 

Motor Driver | Connected | Enabled | Error |
          04 |         1 |       0 |     1 | 

         |   lost   |   sent   | loss ratio | histogram
Commands |        0 |     7799 |       0.00 | 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
Sensors  |        0 |     7803 |       0.00 | 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
```
What does it mean, I checked my I,A,B impulses, they are okey ?

MaximilienNaveau, 2023-04-18:
In this specific case your computer could not connect properly to master board sdk, hence the 0 everywhere.
The prints are unfortunate in your case as they hide the potential issue.
Maybe try to look into the example code and remove the print in order to display the earliest error.

MaximilienNaveau, 2023-04-18:
> sudo PYTHONPATH=. python3 example/listener.pyc -i enp1s0

Furthermore I would not use the .pyc but the py directly to prevent code change/execution mistake