open-dynamic-robot-initiative/master-board #162: Master board doesn't send ack to PC

RainJCloude, 2024-06-05:
Hello, I am having a problem very similar to that in https://github.com/open-dynamic-robot-initiative/master-board/issues/125#issue-1580026324,  however I've no found any solution to my problem yet.

I've followed all the documentation to flash the master-board. So I was able to compile the code of the firmware and flash the masterboard. So far so good.

The problem start when I try to execute the demo.cpp in the sdk folder. In that case the 'Timeout while waiting for ack' appears. 
I'm pretty sure that in that case, the master-board has received the init pack, but it's not able to send the ack package because the blue LED fades.

A similar error occurs if I run the ethernet package in the example folder of the esp-idf package. 

I am quite sure that the problem is related to some kind of configuration that i mess. Maybe there is something that i should set with ``` make menuconfig ``` ??

Moreover, I've followed also the guide on the sdk README for the ethernet communication:
```
sudo ifconfig MY_INTERFACE -arp
``` 

Where MY_INTERFACE is enp3s0, and I've disabled the IPV6 traffic still following the guide in the README

Maybe there is some problem due to the fact that if I type ifconfig in the shell, it appears:
```
enp3s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 04:7c:16:32:89:29  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Thanks a lot for taking your time to help me

thomasfla, 2024-06-05:
Hi, 

The default configuration (mvia make menuconfig) enable ethernet, so your issue is probably related to something else.

Can you make sure that both the sdk and the firmware are in sync (latest commit)?

Best,
Thomas.

RainJCloude, 2024-06-05:
the sdk has not its latest commit, because the README in the master-board packages asked to bring it to a previous commit:
```
cd ~/esp/esp-idf
git checkout 8d1a9c0 # Need to checkout this old version for now
git submodule update --init --recursive
```
About the "make menuconfig" I am not sure hot to enable Ethernet, because in the interface it allows to me to set only the size of the sending packages.

thomasfla, 2024-06-05:
I'm not talking about the ESD IDF SDK, but the sdk of masterboard (the library that will manage the communication with the masterboard. This needs to be the latest commit of this repo.
In the same time, the firmware flashed on the masterboard needs also to be compiled and flash from the latest commit of this repo.

master-board/sdk -> The lib
master-board/firmware -> The firmware (that needs to checkout an old commit of esp-idf to be compiled)

You don't have to change anything in the make menuconfig to have ethernet enabled.

RainJCloude, 2024-06-11:
Okay, sorry I get it. However bot the commit of the firmware and the sdk are at their last commit. The error that appears is still the same:

```
- Main --
if_name: enp5s0f1
Using Ethernet (enp5s0f1)
Shutting down connection (enp5s0f1)
Timeout while waiting for ack.
Masterboard timeout detected. Either the masterboard has been shut down or there has been a connection issue with the cable/wifi.
```
And I've got it starting the sdk :

```
root@cloude-N551JK:~/master-board/sdk/master_board_sdk/bin# ./example enp5s0f1
/example enp5s0f1
```

thomasfla, 2024-06-11:
Can you connect your programmer to the masterbord and give the output of "make monitor" while you start the program?
```
cd master-board/firmware
source setup_esp_idf.bash
make monitor
```

RainJCloude, 2024-06-11:
It outputs this:
[image]

kirillin, 2025-09-01:
I use ethernet and didn't touch configs. 
It work for me, but whan I try to change imu sensor got same issue. For example, 

```
root@fa16d2f41959:/workspace/firmware# make monitor
Toolchain path: /opt/esp/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: esp-2020r3
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.4.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from /opt/esp/idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/ttyUSB0 2000000 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
AU�����5Rڴښ�j%�B�J␦��}�х����х�ѥ��������ձ�Ɂz��
AA�AU���j5R�ETH/WIFI init size 4
ETH/WIFI command size 172
ETH/WIFI ack size 5
ETH/WIFI sensor size 208
SPI size 34
I (59) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (59) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (59) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (59) gpio: GPIO[16]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (69) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (79) emac: emac reset done
I (79) Direct_Ethernet: Ethernet Started
I (79) wifi: wifi driver task: 3ffca79c, prio:23, stack:3584, core=1
I (79) wifi: wifi firmware version: 7c00966
I (79) wifi: config NVS flash: enabled
I (79) wifi: config nano formating: disabled
I (79) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (79) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (89) wifi: Init dynamic tx buffer num: 32
I (99) wifi: Init data frame dynamic rx buffer num: 32
I (99) wifi: Init management frame dynamic rx buffer num: 32
I (99) wifi: Init management short buffer num: 32
I (99) wifi: Init static rx buffer size: 1600
I (99) wifi: Init static rx buffer num: 10
I (99) wifi: Init dynamic rx buffer num: 32
I (179) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (179) wifi: mode : sta (a8:03:2a:d7:66:68)
I (179) wifi: set country: cc=JP schan=1 nchan=14 policy=1

I (179) ESPNOW: espnow [version: 1.0] init
initialise IMU
WIT IMU initializing... 
UART OK: Connection established, sent 4 bytes, received 0 bytes
Setup done
I (4079) Direct_Ethernet: Ethernet Link Up
I (4079) Direct_Ethernet: Ethernet HW Addr a8:03:2a:d7:66:6b
I (4079) Direct_Ethernet: Ethernet got IP
I (4079) Direct_Ethernet: TCP/IP initialization finished.
I (4079) Direct_Ethernet: TCP|IP 	 IP:192.168.1.0
I (4079) Direct_Ethernet: TCP|IP 	 MASK:255.255.255.0
I (4079) Direct_Ethernet: TCP|IP 	 GW:192.168.1.1
I (4079) tcpip_adapter: eth ip: 192.168.1.0, mask: 255.255.255.0, gw: 192.168.1.1
I (4639) ESPNOW: espnow [version: 1.0] deinit
I (4639) wifi: flush txq
I (4639) wifi: stop sw txq
E (4739) Direct_Ethernet: Error occurred while sending eth frame: error code 0x101
E (4749) Direct_Ethernet: Error occurred while sending eth frame: error code 0x101
E (4759) Direct_Ethernet: Error occurred while sending eth frame: error code 0x101
E (4769) Direct_Ethernet: Error occurred while sending eth frame: error code 0x101
// and many many those rows next
```